USST-RADSAT-SK/software-and-command #25: UART Wrapper

tyrelkostyk, 2021-04-06:
Implement a simple wrapper around the UART driver provided in the HAL

ThomasSlotboom, 2021-05-08:
Deadline - June 1

jacobwaskowic, 2021-05-11:
Contains three functions for now:
UART_blockingTransmit():
Inputs:
- Takes in a pointer to data to be sent over UART
- Takes in the size in bytes to be sent
Outputs:
- An integer error code, 0 indicates success. Non-zero indicates a failure, which is specified in UART.h

UART_blockingRecieve():
Inputs:
- Takes in a pointer to a buffer to receive data into from UART
- Takes in the size in bytes to be received
Outputs:
- An integer error code, 0 indicates success. Non-zero indicates a failure, which is specified in UART.h

UART_init():
Initializes UART and enables receiving. Should be ran at the beginning of our main application once.

jacobwaskowic, 2021-05-11:
Note that these functions are blocking. At some point later on, an additional task to manage these functions would be a good idea.

closed 2021-07-09 by ThomasSlotboom