USST-RADSAT-SK/software-and-command #31: Implement FRAM Wrapper

tyrelkostyk, 2021-04-06:


ThomasSlotboom, 2021-05-08:
Deadline - June 1

tyrelkostyk, 2021-05-11:
Deadline of June 1 sounds reasonable. It will be basic and likely untested on hardware, however.

Some early thoughts on an FRAM wrapper implementation:

**Public API**
- framInit( )
    - calls the HAL's `FRAM_start` to init the driver
    - calls the HAL's `FRAM_getDeviceID` and `FRAM_getMaxAddress` to populate local static variables that may be used later
    - deals with any block protection (if necessary)
- framWrite( )
    - Writes a buffer of data to FRAM. Uses `FRAM_writeAndVerify`
- framRead( )
    - Uses `FRAM_read`

**Some Private functions**
- framMaxAddress( )
    - Used to ensure we're not writing out of bounds

closed 2021-12-12 by ChrisAmaya