USST-RADSAT-SK/software-and-command #37: Fix Launch Configuration

tyrelkostyk, 2021-05-04:
Currently unable to program or Debug the Dev Kits due to an issue regarding the provided Launch configurations. Have contacted ISISpace and gotten a response, but may need to follow up again

tyrelkostyk, 2021-05-04:
Made some progress, getting the following output now (instead of just nothing like before).

```
symbol-file C:\\Users\\short\\Documents\\git-repos\\software-and-command\\radsat-sk\\Debug\\radsat-sk.elf
target remote localhost:2331
localhost:2331: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.
monitor flash breakpoints = 0
"monitor" command not supported by this target.
monitor flash download = 0
"monitor" command not supported by this target.
monitor AllowSimulation = 0
"monitor" command not supported by this target.
monitor endian little
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEF1C = 0x00020000 // Set EBI Slew-Rate to slow and voltage to 1.8V
"monitor" command not supported by this target.
monitor sleep 10
"monitor" command not supported by this target.
monitor writeu32 0xFFFFFC20 = 0x00004001 // Enable main oscillator
"monitor" command not supported by this target.
monitor sleep 10
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEF1C = 0x00020000 // Set EBI Slew-Rate to slow and voltage to 1.8V
"monitor" command not supported by this target.
monitor sleep 10
"monitor" command not supported by this target.
monitor writeu32 0xFFFFFC28 = 0x202A3F01 // Set PLLA to 800MHz
"monitor" command not supported by this target.
monitor sleep 10
"monitor" command not supported by this target.
monitor writeu32 0xFFFFFC2C = 0x10193F05 // Set PLLB for USB usage
"monitor" command not supported by this target.
monitor sleep 10
"monitor" command not supported by this target.
monitor writeu32 0xFFFFFC30 = 0x00001300 // Select prescaler
"monitor" command not supported by this target.
monitor sleep 10
"monitor" command not supported by this target.
monitor writeu32 0xFFFFFC30 = 0x00001302 // Select master clock
"monitor" command not supported by this target.
monitor sleep 10
"monitor" command not supported by this target.
echo Master clock ok.\n
Master clock ok.
echo Configuring the SDRAM controller...\n
Configuring the SDRAM controller...
monitor writeu32 0xFFFFEF1C = 0x2 // Enable EBI chip select for the SDRAM 
"monitor" command not supported by this target.
monitor writeu32 0xFFFFF870 = 0xFFFF0000 // Enable PC16-PC31 pins
"monitor" command not supported by this target.
monitor writeu32 0xFFFFF874 = 0x00000000 // Enable PC16-PC31 pins
"monitor" command not supported by this target.
monitor writeu32 0xFFFFF804 = 0xFFFF0000 // Enable PC16-PC31 pins
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA08 = 0xF7339375 // SDRAM configuration for 32MB: Change this one if the SDRAM Size changes
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x1
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x2
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x4
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x4
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x4
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x4
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x4
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x4
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x4
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x4
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x3
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA00 = 0x0
"monitor" command not supported by this target.
monitor writeu32 0x20000000 = 0
"monitor" command not supported by this target.
monitor writeu32 0xFFFFEA04 = 0x39D
"monitor" command not supported by this target.
echo SDRAM configuration ok.\n
SDRAM configuration ok.
monitor flash breakpoints = 0
"monitor" command not supported by this target.
monitor flash download = 0
"monitor" command not supported by this target.
monitor AllowSimulation = 0
"monitor" command not supported by this target.
load C:\\Users\\short\\Documents\\git-repos\\software-and-command\\radsat-sk\\Debug\\radsat-sk.elf 
You can't do that when your target is `None'
continue
The program is not being run.
kill
The program is not being run.
```

ThomasSlotboom, 2021-05-08:
Deadline - June 1

closed 2022-03-27 by matthew-buglass